fsphil/hacktv #120: Support the TBC (4fsc) CVBS/S-Video signal file format

harrypm, 2024-10-07:
Notation to add support for the .tbc formats NTSC and PAL.

These are files are mainly genarated by the [decode projects](https://github.com/oyvindln/vhs-decode/wiki).

The file format can be genarated in software with digital image assets or video assets details are on the [chroma-encoder page](https://github.com/happycube/ld-decode/wiki/ld-chroma-encoder) or from public FM RF archives of composite/colour-under tape formats and LaserDiscs linked on the wiki's.

We currently use the [FL2K Method](https://github.com/oyvindln/vhs-decode/wiki/TBC-To-Analogue)  but its far from ideal seeing matts tv barn [video](https://www.youtube.com/watch?v=0vaM3o4v_2o) on generating broadcast complient CVBS or well as close as possible to it has got me thinking about the future of playout in hardware again, and getting tbc support implimented into hacktv seems like a good step to make it more flexible to do so.

fsphil, 2024-11-24:
I tried creating a sample .tbc file with:

`ffmpeg -i INPUT.mkv -an -vf "scale=922:576,pad=922:576:0:3" -vcodec rawvideo -pix_fmt yuv444p16 -f rawvideo - | ld-chroma-encoder -p yuv -f PAL - OUTPUT.tbc`

This produces a file with 1135 samples per line, but it doesn't seem to show the expected shift each line due to the 0.0064 rounding error. Is that expected?

harrypm, 2024-11-24:
I believe that is expected, I'm not 100% sure off the top of my head though would have to confirm with @happycube.

fsphil, 2025-01-22:
I've added an `--s-video` argument, this works for file and fl2k outputs.
For files, luminance and chrominance samples are interleaved.

happycube, 2025-01-25:
Yeah... I didn't want to apply another horizontal scaling after the TBC phase, and figured it was Close Enough.  Maybe it isn't really, but... *shrug*

fsphil, 2025-01-25:
> Yeah... I didn't want to apply another horizontal scaling after the TBC phase, and figured it was Close Enough. Maybe it isn't really, but... _shrug_

So that's just a limitation of `ld-chroma-encoder` rather than something in the TBC file format?

fsphil, 2025-03-13:
Note for myself: Probably best to treat PAL TBC files as 17734375 Hz (1135 * 25 * 625) rather than true 4fsc (17734475 Hz).