fsphil/hacktv #121: No sound on DMAC or NICAM

EdReardon, 2025-01-02:
Odd one,

I have no sound output on DMAC nor a valid NICAM output using either the command line or the HackTV GUI.

Am I missing a dependency or is this a known issue,  my DMAC is entirely silent and neither of my NICAM decoders will receive a NICAM signal from the HackRF and stay completely silent.  Normal FM +6MHz is being transmitted and received fine.

The PC has more than enough juice, the picture is perfect and I'm not dropping any frames, just seems odd I have neither DMAC audio or NICAM audio?

Thanks

captainjack64, 2025-01-02:
Are you seeing MAC information on your decoder/receiver when in DMAC mode? What decoder is it?

EdReardon, 2025-01-02:
It’s a very late ITT BSB box, however if I do mono at 16kHZ it works, that doesn’t bother me, it’s only a mono studio monitor anyway.

The NICAM flat refuses to work though, neither of my NIcAM decoders will see NICAM from PAL, looking at the code it looks like simple bit-shifting with no dependency for NICAM so I don’t think it’s a dependency issue?

fsphil, 2025-01-02:
I did a quick test with System I and the NICAM audio is coming through fine on my TV.
This is with the latest code on my branch (hacktv 20241229-801b2f6) and the following command line:

`$ ./hacktv -f 735.25e6 -g 30 --filter -m i test`

Can you let us know with version you have and what command line you're using?

EdReardon, 2025-01-03:
> I did a quick test with System I and the NICAM audio is coming through fine on my TV. This is with the latest code on my branch (hacktv 20241229-801b2f6) and the following command line:
>
> `$ ./hacktv [...]

Thanks,  tried that command-line and nothing on either of my Arcam NICAM decoders, unless they're especially fussy etc.  I have normal program sound though on the FM sub-carrier.  I only set this up yesterday so running  20241229-801b2f6 on Ubuntu 24.

Unless I have two faulty NICAM decoders, but that seems unlikely?  Might just be "one of those things?"

fsphil, 2025-01-03:
That it also doesn't work on the BSB box suggests it's not a problem with the Arcam. It feels more like a bug in the encoder.

What does `gcc -v` show?

EdReardon, 2025-01-03:
> That it also doesn't work on the BSB box suggests it's not a problem with the Arcam. It feels more like a bug in the encoder.
> 
> What does `gcc -v` show?

Hi there,

This is the output from the compiler:

gcc -v

`Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-linux-gnu/13/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 13.3.0-6ubuntu2~24.04' --with-bugurl=file:///usr/share/doc/gcc-13/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-13 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/libexec --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-libstdcxx-backtrace --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --enable-cet --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-13-fG75Ri/gcc-13-13.3.0/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-13-fG75Ri/gcc-13-13.3.0/debian/tmp-gcn/usr --enable-offload-defaulted --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-serialization=2
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 13.3.0 (Ubuntu 13.3.0-6ubuntu2~24.04) 
`

fsphil, 2025-01-03:
I tested it with the Ubuntu 24.04.1 LTS + Ubuntu 24.10 live images and NICAM is working fine there.
Are you able to do the same test? In both cases I booted, opened the terminal and ran exactly this:
```
$ sudo apt update
$ sudo apt install git gcc make pkg-config libavcodec-dev libavformat-dev libavdevice-dev libswscale-dev libhackrf-dev
[plugged in the hackrf]
$ git clone https://github.com/fsphil/hacktv
$ cd hacktv/src
$ make
$ ./hacktv -f 735.25e6 -g 30 --filter -m i test
```

EdReardon, 2025-01-03:
Hi,  well I've cleaned the whole HackTV install and installed all the dependencies,  I was missing libhacktv-dev and I thought that may have cracked it but alas,  still two solid 'red' NICAM indicators on both Arcam units.

`gcc -g -Wall -pthread -O3  -DVERSION=\"20241229-801b2f6\" -DHAVE_HACKRF -DHAVE_SOAPYSDR -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/usr/local/include/libhackrf -I/usr/include/libusb-1.0  -c hacktv.c -o hacktv.o
gcc -g -Wall -pthread -O3  -DVERSION=\"20241229-801b2f6\" -DHAVE_HACKRF -DHAVE_SOAPYSDR -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/usr/local/include/libhackrf -I/usr/include/libusb-1.0  -c common.c -o common.o
gcc -g -Wall -pthread -O3  -DVERSION=\"20241229-801b2f6\" -DHAVE_HACKRF -DHAVE_SOAPYSDR -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/usr/local/include/libhackrf -I/usr/include/libusb-1.0  -c fir.c -o fir.o
gcc -g -Wall -pthread -O3  -DVERSION=\"20241229-801b2f6\" -DHAVE_HACKRF -DHAVE_SOAPYSDR -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/usr/local/include/libhackrf -I/usr/include/libusb-1.0  -c vbidata.c -o vbidata.o
gcc -g -Wall -pthread -O3  -DVERSION=\"20241229-801b2f6\" -DHAVE_HACKRF -DHAVE_SOAPYSDR -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/usr/local/include/libhackrf -I/usr/include/libusb-1.0  -c teletext.c -o teletext.o
gcc -g -Wall -pthread -O3  -DVERSION=\"20241229-801b2f6\" -DHAVE_HACKRF -DHAVE_SOAPYSDR -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/usr/local/include/libhackrf -I/usr/include/libusb-1.0  -c wss.c -o wss.o
gcc -g -Wall -pthread -O3  -DVERSION=\"20241229-801b2f6\" -DHAVE_HACKRF -DHAVE_SOAPYSDR -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/usr/local/include/libhackrf -I/usr/include/libusb-1.0  -c video.c -o video.o
video.c: In function ‘_vid_passthru_process’:
video.c:3564:9: warning: ignoring return value of ‘fread’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
 3564 |         fread(s->passline, sizeof(int16_t) * 2, l->width, s->passthru);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gcc -g -Wall -pthread -O3  -DVERSION=\"20241229-801b2f6\" -DHAVE_HACKRF -DHAVE_SOAPYSDR -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/usr/local/include/libhackrf -I/usr/include/libusb-1.0  -c fifo.c -o fifo.o
gcc -g -Wall -pthread -O3  -DVERSION=\"20241229-801b2f6\" -DHAVE_HACKRF -DHAVE_SOAPYSDR -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/usr/local/include/libhackrf -I/usr/include/libusb-1.0  -c mac.c -o mac.o
In function ‘_bits’,
    inlined from ‘_line_625’ at mac.c:1464:7,
    inlined from ‘mac_next_line’ at mac.c:1687:7:
mac.c:369:34: warning: ‘df’ may be used uninitialized [-Wmaybe-uninitialized]
  369 |                 if(bits & 1) data[offset >> 3] |= b;
      |                              ~~~~^~~~~~~~~~~~~
mac.c: In function ‘mac_next_line’:
mac.c:1450:17: note: ‘df’ declared here
 1450 |         uint8_t df[16];
      |                 ^~
In function ‘_bits’,
    inlined from ‘_line_625’ at mac.c:1464:7,
    inlined from ‘mac_next_line’ at mac.c:1687:7:
mac.c:369:34: warning: ‘df’ may be used uninitialized [-Wmaybe-uninitialized]
  369 |                 if(bits & 1) data[offset >> 3] |= b;
      |                              ~~~~^~~~~~~~~~~~~
mac.c: In function ‘mac_next_line’:
mac.c:1450:17: note: ‘df’ declared here
 1450 |         uint8_t df[16];
      |                 ^~
In function ‘_bits’,
    inlined from ‘_line_625’ at mac.c:1479:7,
    inlined from ‘mac_next_line’ at mac.c:1687:7:
mac.c:369:34: warning: ‘df’ may be used uninitialized [-Wmaybe-uninitialized]
  369 |                 if(bits & 1) data[offset >> 3] |= b;
      |                              ~~~~^~~~~~~~~~~~~
mac.c: In function ‘mac_next_line’:
mac.c:1450:17: note: ‘df’ declared here
 1450 |         uint8_t df[16];
      |                 ^~
gcc -g -Wall -pthread -O3  -DVERSION=\"20241229-801b2f6\" -DHAVE_HACKRF -DHAVE_SOAPYSDR -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/usr/local/include/libhackrf -I/usr/include/libusb-1.0  -c dance.c -o dance.o
gcc -g -Wall -pthread -O3  -DVERSION=\"20241229-801b2f6\" -DHAVE_HACKRF -DHAVE_SOAPYSDR -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/usr/local/include/libhackrf -I/usr/include/libusb-1.0  -c eurocrypt.c -o eurocrypt.o
gcc -g -Wall -pthread -O3  -DVERSION=\"20241229-801b2f6\" -DHAVE_HACKRF -DHAVE_SOAPYSDR -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/usr/local/include/libhackrf -I/usr/include/libusb-1.0  -c videocrypt.c -o videocrypt.o
gcc -g -Wall -pthread -O3  -DVERSION=\"20241229-801b2f6\" -DHAVE_HACKRF -DHAVE_SOAPYSDR -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/usr/local/include/libhackrf -I/usr/include/libusb-1.0  -c videocrypts.c -o videocrypts.o
gcc -g -Wall -pthread -O3  -DVERSION=\"20241229-801b2f6\" -DHAVE_HACKRF -DHAVE_SOAPYSDR -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/usr/local/include/libhackrf -I/usr/include/libusb-1.0  -c syster.c -o syster.o
gcc -g -Wall -pthread -O3  -DVERSION=\"20241229-801b2f6\" -DHAVE_HACKRF -DHAVE_SOAPYSDR -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/usr/local/include/libhackrf -I/usr/include/libusb-1.0  -c acp.c -o acp.o
gcc -g -Wall -pthread -O3  -DVERSION=\"20241229-801b2f6\" -DHAVE_HACKRF -DHAVE_SOAPYSDR -I/usr/include/x86_64-linux-gnu -I/usr/local/include -I/usr/local/include/libhackrf -I/usr/include/libusb-1.0  -c vits.c -o vits.o
[... cut ...]
gcc -o hacktv hacktv.o common.o fir.o vbidata.o teletext.o wss.o video.o fifo.o mac.o dance.o eurocrypt.o videocrypt.o videocrypts.o syster.o acp.o vits.o vitc.o nicam728.o sis.o av.o av_test.o av_ffmpeg.o rf.o rf_file.o rf_hackrf.o rf_soapysdr.o -g -lm -pthread  -lavcodec -lavformat -lavdevice -lswscale -lswresample -lavutil -L/usr/local/lib -lhackrf -lSoapySDR 
~/hacktv/src$ make install
`

Totally at a loss now, great sharp picture and good solid program FM audio but these boxes just won't pick up the NICAM :(

Thank you so much for taking the time to look in to this! :)

fsphil, 2025-01-04:
If you're able to can you try applying the attached patch. It will dump the raw NICAM frames to a file `nicam.dump` each time you run hacktv. Run the following command for a few seconds:

`./hacktv -o /dev/null -m i test`

Then send me the `nicam.dump` file that produces. I want to compare it with the output from my copy.

[nicamdump.patch.zip](https://github.com/user-attachments/files/18306875/nicamdump.patch.zip)

EdReardon, 2025-01-04:
HI mate,

This is my ouptut post-patch of `./hacktv -o /dev/null -m i test`
[nicamdump.zip](https://github.com/user-attachments/files/18307161/nicamdump.zip)

Thanks :)

fsphil, 2025-01-04:
That looks fine. Here's another test, try downloading this file: https://www.sanslogic.co.uk/hacktv/pal-i-sample.zip

This contains a ~10 second sample of a PAL I signal, I've verified it works on my NICAM receivers. You can transmit it directly through the hackrf with this command:

`hackrf_transfer -t pal-i-sample.iq8 -f 735.25e6 -x 30 -s 16000000`

Change frequency and gain to whatever you need.

EdReardon, 2025-01-04:
I'm afraid I've still got two solid red LEDs.

I'll have to chalk this up to equipment failure :( Annoying that!

Thank you so very much for taking the time to look in to this though.

fsphil, 2025-01-04:
Yeah as unlikely as it is, it does seem like a problem with your receiver. Have you any other devices to test it with?

EdReardon, 2025-01-04:
I have an early 2000s flatscreen of some description in the attic, I’ll get it down tomorrow.

Odd question, is there a way to dump a pcm-wav as a NICAM represention in the software then expand it back to 14bit PCM to at least emulate the “sound” of NICAM? 

I’m thinking at least I could do A/B with 16/44.1 & NICAM?

I appreciate that’s probably out of scope!

fsphil, 2025-01-04:
I built a NICAM receiver in GNU Radio a while back, it uses an rtl-sdr to do the actual reception if you have one of those.
Probably wouldn't be too hard to modify that to work on a raw hacktv output directly. I'll see if I can dig that out.

fsphil, 2025-01-04:
Actually this is probably easier, it's an old python script that can take the NICAM dump files produced by that earlier patch:

```
$ ./nicam-dec.py nicam.dump test.pcm # Decode
$ aplay -fcd -r32000 test.pcm # Playback
$ ./nicam-dec.py nicam.dump /dev/stdout | aplay -fcd -r32000 - # Decode and playback at the same time
```

[nicam-dec.py.gz](https://github.com/user-attachments/files/18308473/nicam-dec.py.gz)

EdReardon, 2025-01-06:
Sir that's amazing!

Finally had a chance to play with it, that's absolutely what I was looking for - and it's saved me quite a bit of time trying to write my own! I've used ffmpeg to convert the outputs to wav!

Thanks again,  keep an eye out on YouTube in the coming week or so about an Arcam NICAM decoder :)

Cheers

fsphil, 2025-01-06:
No worries. I just noticed the script isn't apply de-emphasis so the output may sound a bit thin. I'll update the script.

EdReardon, 2025-01-06:
> No worries. I just noticed the script isn't apply de-emphasis so the output may sound a bit thin. I'll update the script.

Ah, you read my mind! If you could that would be most helpful! I've just spend half an hour fiddling with my soundcard drivers before noticing that too!

Thanks :)

fsphil, 2025-01-07:
Try this one. The filter is a bit slow in python, it doesn't run in real time on my laptop now.
[nicam-dec.py.gz](https://github.com/user-attachments/files/18325955/nicam-dec.py.gz)

EdReardon, 2025-01-07:
Sir, that's amazing! I was fiddling trying to do it in C, but that's saved me some tremendous effort!

Don't worry about it being slow, I don't need to do it in real time, I just need it to spit out wav files.

Thanks again sir!

fsphil, 2025-01-09:
Update: I've dug out a D/D2-MAC decoder and yep, no MAC audio except on medium quality / 16kHz mode. So it *is* a bug!

fsphil, 2025-01-09:
That might fix it and/or break something else. If you get a moment can you test please.

EdReardon, 2025-01-13:
Hi Phil,

That's got the sound working on a BSB box, although it does seem to cause a few more picture drop outs (appears to occur on transients,  beats in music for example) but might just be coincidence.

No reported dropped frames, but I'm definitely convinced there is more sporadic picture roll and drop-out since this update?  Does that make any sense?

fsphil, 2025-01-13:
I'm saw something similar but I blamed it on some hardware issues I've been having with my hackrf.
Out of curiosity, what set-up are you using to transmit to the BSB box? Direct to IF or using a Ku-band harmonic via LNB?

EdReardon, 2025-01-13:
Yes I’m fairly certain it’s high amplitude audio encroaching on to the video signal. I do get a loss of sync’ during some beats.

I’m using IF by some antenuation and a matching transformer.

fsphil, 2025-01-13:
Are there any changes in the video that match the beat? Fast changes in contrast can sometimes cause D/D2-MAC decoders to glitch.

fsphil, 2025-01-14:
I've fixed another bug introduced in the recent FIR filter rearranging. I'm not sure if it would affect D/D2-MAC but are you able to test?

closed 2025-04-03 by fsphil