fsphil/hacktv #122: Oak Orion/Delta/Cablecrypt/SSavi Scrambling

paul19920801, 2025-03-28:
Hi all - I was wondering if support potentially could be added for scrambling which used a combination of oak orion/ssavi scrambling. 
 
Mnet used this

[IRDETO DELTA DECODER.pdf](https://github.com/user-attachments/files/19512899/IRDETO.DELTA.DECODER.pdf)
[info_irdeto.pdf](https://github.com/user-attachments/files/19512897/info_irdeto.pdf)
[Descrambling-1.pdf](https://github.com/user-attachments/files/19512895/Descrambling-1.pdf)
[European Scrambling Systems_text.pdf](https://github.com/user-attachments/files/19512898/European.Scrambling.Systems_text.pdf)
[1984-video-scrambling-an-overview.pdf](https://github.com/user-attachments/files/19512896/1984-video-scrambling-an-overview.pdf)
[M-NET decoder production a technical analysis.pdf](https://github.com/user-attachments/files/19512894/M-NET.decoder.production.a.technical.analysis.pdf)

I attempted a plugin, and got the picture close - but was getting noise all over the FM carrier - and figured this would be best left to the experts. 

If anyone would be willing to look into this - I have guides I can supply which have in-dept information as to how the system worked.  Apparantly there is enough info - to come fairly close. 

Thanks in advance

captainjack64, 2025-03-29:
And Luxcrypt, which is similar to Irdeto but apparently uses a different frequency of ~5.72MHz in place of sync pulses.

paul19920801, 2025-03-29:
> And Luxcrypt, which is similar to Irdeto but apparently uses a different frequency of ~5.72MHz in place of sync pulses.

Indeed - I hope the documentation I provided will be enough to implement support...

paul19920801, 2025-03-30:
If a decoder is needed for testing - a friend of mine would willing to contribute to the cause, and donate an M-net decoder - provided shipping is paid for..

fsphil, 2025-03-30:
Was the digital audio part ever documented? That may be the trickiest part to get working.

paul19920801, 2025-03-30:
> Was the digital audio part ever documented? That may be the trickiest part to get working.

And also a lot of providers I don't think use the sound protection I think it was mainly in South Africa with mnet that scrambled the sound

fsphil, 2025-03-30:
Would it be possible to get a recording of M-NET on VHS, or at least a dump of the VBI data from a recording?

paul19920801, 2025-03-30:
> Would it be possible to get a recording of M-NET on VHS, or at least a dump of the VBI data from a recording?

https://www.youtube.com/watch?v=EBhixLqUbWM

This was capped by a friend of mine - M-net's scrambled picture. He's intrested in this kind of thing - if this isn't enough let me know - and will look for more.

fsphil, 2025-03-30:
It would need to be a direct capture. It's so I can see if there is any data in the VBI to control the decoder.

Though that video is interesting, it seems like there are more than the 3 bytes (24-bits) that some of those PDF documents mention.

paul19920801, 2025-03-30:
> It would need to be a direct capture. It's so I can see if there is any data in the VBI to control the decoder.
> 
> Though that video is interesting, it seems like there are more than the 3 bytes (24-bits) that some of those PDF documents mention.

Will see what I can do - spoke to my friend & contact in South Africa - hopeful he'll also post here - as he knows more about this than I do.

paul19920801, 2025-03-31:
in the mean time here are more patients. These were filed by Pieter den Toonder - the founder of Irdeto

https://patents.google.com/patent/CA1150818A/
https://patents.google.com/patent/US4353088A
https://patents.google.com/patent/US4447828A
https://patents.google.com/patent/US4568974A
https://patents.google.com/patent/US4340906A