fsphil/hacktv #124: Audio level low

VasylSamoilov, 2025-04-27:
I was experimenting with different receivers, and it seems to me that encoded audio level is very low (to be specific, 50% of max range). 
I have a usb tv tuner that is capable of showing oscilloscope representation of single scan line and audio levels, and even in test generation mode beeps are at 50% max level. All other receivers I have are also seem a little bit quiet, with either test pattern of file playback. 
Am I doing something wrong? Is there a way to make audio louder?

fsphil, 2025-05-06:
You can use `--volume` to amplify the audio, but using this may cause clipping.

Low volume has been mentioned before but I believe all the various audio modulators (AM/FM) are using correct levels. Is it possible your audio source is just quiet?

VasylSamoilov, 2025-05-06:
@fsphil  I'm using build-in test pattern as a reference, and according to my tuner it demodulates test pattern audio tones exactly to 50% of possible maximum audio level. Maybe there is something wrong with audio FM deviation calculation? I'm mostly testing PAL B/G (that is what my tuner is capable of debugging), so I can't tell if everything is affected.

VasylSamoilov, 2025-05-06:
I have encountered similar issue when debugging osmocom-analog 1G AMPS cell network base station emulator, which uses same FM modulation for voice channels. 
The catch is that peak frequency deviation should happen at expected **nominal volume level** for analog devices. Nominal volume level is 50% or less of max possible amplitude for digital recordings (which is -3 dBm).

see https://gitea.osmocom.org/cellular-infrastructure/osmocom-analog/issues/11 if you are interested. Fun fact, it's due to AMPS networks many FM receivers including TVs where prohibited to receive around 850 MHz range, because they could snoop on analog voice channels of 1G networks :)

VasylSamoilov, 2025-05-30:
I done some more testing, volume parameter value of 2 seems to produce correct level on the receiver (when using test pattern).
So, for anyone looking into the issue in the future - remember to look into correct correlation between levels of analog and digital systems and their naming - peak FM deviation of audio signal should happen at nominal volume level, which for digital recordings is 50% max amplitude or -3 dBm. It would be nice to have volume parameter of 2 as default.