fsphil/hacktv #125: SWScaler error when setting sample rate for NTSC output.

DMNewb2020, 2025-05-17:
I'm attempting to turn a still image into a NTSC video. I've started by using ffmpeg to turn the still image (which is 640x480 in dimension) into a MP4 using the following:

ffmpeg -loop 1 -i amateur-radio-NTSC-format.jpg -t 10 -r 30 testVideoNtsc.mp4

Then I'm attempting to use hacktv to turn this into a baseband NTSC signal using the following:

hacktv -o ntscVideo-13M5Fs.rs16 -m ntsc -s 13.5e6 testVideoNtsc.mp4

What I get when I attempt to run this command is (my emphasis added): 

Warning: Pixel rate 13 may not work well with this mode.
Next valid pixel rates: 0, 4500000
Video: 0x480 29.97 fps (full frame 0x525)
Sample rate: 13
Opening 'testVideoNtsc.mp4'...
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'testVideoNtsc.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.76.100
  Duration: 00:00:10.00, start: 0.000000, bitrate: 76 kb/s
  Stream #0:0(und): Video: h264 (High 4:4:4 Predictive) (avc1 / 0x31637661), yuvj444p(pc, bt470bg/unknown/unknown), 640x480 [SAR 1:1 DAR 4:3], 73 kb/s, 30 fps, 30 tbr, 15360 tbn, 60 tbc (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
Using video stream 0.
[swscaler @ 0x5600be3853c0] deprecated pixel format used, make sure you did set range correctly
**[swscaler @ 0x5600be3853c0] 640x480 -> 0x480 is invalid scaling dimension**

However, when I run the command without the sample rate option (as follows): hacktv -o ntscVideo-13M5Fs.rs16 -m ntsc testVideoNtsc.mp4

I get a usable NTSC output, although it scrolls when I view it with a Gnu Radio flowgraph I created. I imagine this is due to the trying to use 16 MHz sample rate (rather than 13.5 or 18, which would be an integer multiple of the 4.5 MHz NTSC wants).

Any ideas?

fsphil, 2025-05-17:
The sample and pixel rate arguments are expected to be integers, eg. `-s 13500000` rather than `-s 13.5e6`. It's parsing the latter as `13`.

It's better if it worked both ways so I've just pushed an update that fixes this.

DMNewb2020, 2025-05-18:
Got it! Thank you!

closed 2025-05-18 by fsphil