fsphil/hacktv #127: FFMPEG having trouble with audio

RCCRAFT1, 2025-06-03:
So I recently decided to try this out with my HackRF, and I noticed something while trying to play a video source.

Videos with no sound in AV1 format work fine, however if I try to add a video with sound, hacktv returns with this:

--
Metadata: |  
handler_name | : VideoHandLer
vendor_id : (•] (0] (0) (0)
  |  
Metadata:
handler_name | : SoundHandler
vendor_id | : (0) (0) (0) (0]
Using video stream 0 |  
Using audio stream 1, |  
[SWR @ 0x7f96d17bf000) Input channel layout "" is invalid or unsupported.
Failed to initialise the resampling context

I am unsure of what to do, as I am on MacOS and after doing research, the fix that is needed I can't run on Mac, since apt on Mac doesn't exist. Would I have to continue using this without sound? Or is there a way to get the sound running?

captainjack64, 2025-06-03:
Do you have a copy of the file? Does the file play OK on the Mac using VLC or similar?

ffmpeg can be updated using brew on the Mac, so try that.

RCCRAFT1, 2025-06-03:
I'm using the latest version of ffmpeg, and the file plays just fine.

RCCRAFT1, 2025-06-05:
https://github.com/user-attachments/assets/d34f1e51-4554-46ef-935f-5ed15b208d3b

RCCRAFT1, 2025-06-05:
I realized that after trying the file on my Arch computer. Is it a mac-specific issue?

RCCRAFT1, 2025-06-05:
im using the latest version from brew. should i he using another repo?

fsphil, 2025-06-24:
Can you run this and paste the result: `hacktv --version`

RCCRAFT1, 2025-06-25:
I am running captainjack's build as if i remember correctly, it was the only one that has semi-native mac support

fsphil, 2025-06-25:
Yeah I'm seeing the same issue with that branch.