fsphil/hacktv #131: one more question, where is the help/main page?

concretebracelet, 2025-06-17:
I didn't get to ask that question when you closed that previous issue. I can't find there!!

fsphil, 2025-06-17:
You can still reply to closed issues.
The help text can be displayed by using the `--help` command.

concretebracelet, 2025-06-19:
...how do I do commands?

03stevensmi, 2025-06-20:
> ...how do I do commands?

You know what a terminal is? if not... there is no chance you'll get this working.
Your gonna want to look up a 10 minuet youtube video and learn some really basic commands like i once did 10 years ago.

like "CD, ls (or dir if your on windows) etc.

Here's what i would do if i was you!
Forget all this, forget commands and terminals etc and instead,
use hacktv with this: https://github.com/steeviebops/hacktv-gui/releases

This will let you use hacktv without any commands, learning etc.

All you need to do is download and install this link for windows:
https://github.com/steeviebops/hacktv-gui-installer/releases/download/1.3/hacktv-gui_setup.exe

done!
You can now use hacktv like any other normal software without learning how to use terminal commands.
;)

As for getting a hackrf or fl2k, setting it up... and actually using hacktv...
that's gonna be a HELL lot more harder than leaning a few commands.

but... good news is there is a youtube video you can follow to get it up and hooked up:
https://www.youtube.com/watch?v=g2a7bKlhGd4

Hope this has helped you out a bit Mate!
Cheers!
;D

concretebracelet, 2025-06-20:
I **do** use the HackTV GUI.