fsphil/hacktv #132: What is the QPSK signal in the raw modulated PAL spectrum?

JesterNoFool, 2025-09-22:
I've created a PAL signal with hacktv using the following command: **hacktv -o PAL-modulated-16M0FS.cs16 -m i -s 16000000 input-video.mp4**
When I look at [the spectrum using Gnu Radio](https://www.site2241.net/hacktv/PAL-raw-spectrum-and-constellation-annotated.jpg), I see what appears to be a modulated signal (red arrow) just higher in the spectrum than the audio carrier. When I shift and filter the spectrum of that signal, then look at it with a constellation sink, it appears to be a QPSK signal (see bottom display of linked image). I'm no expert in PAL, so I was wondering if someone could explain what this signal is.

Thanks!

fsphil, 2025-09-22:
That's the NICAM-728 digital audio carrier. It was used to broadcast stereo/multilingual audio in the UK and other countries back in the analogue TV days:

https://en.wikipedia.org/wiki/NICAM#Broadcasts_to_the_public

JesterNoFool, 2025-09-22:
Thank you! Am I reading correctly that a NICAM-728 signal will extend into the next channel higher's video signal? It appears that its 3 dB bandwidth is just at the edge of the channel, but some of the lower energy portion of the signal extends beyond that. Is this correct?

JesterNoFool, 2025-09-22:
Ah! Figured it out! The answer to my question is "Yes, it extends slightly into the next channel.", according to EN 300 163 V1.2.1 (1997-10). Thanks, again, for the education! "hacktv" is awesome!

closed 2025-09-22 by JesterNoFool